mchck/mchck #124: Project status?

beriberikix, 2015-07-08:
Hi all, is this project still active? There hasn't been any activity in a while and the IRC channel was silent. I'd love to buy or group-order a few of these if they ever make it on to Tindie.

beriberikix, 2015-07-08:
Looking at the mailing list it looks like there might be a cheaper board in the works, based off an [M0+ chip](https://0x2c.org/mailman/pipermail/nerds/2015-July/000688.html). I'd love to get my hands on a few :smile:

Jorgen-VikingGod, 2017-06-05:
are there any progress here?
Is it possible to use it with ArduinoIDE and Arduino-libs?

corecode, 2017-06-05:
no, the project goal was to develop a different set of libraries, now called mchck-os.  if you want to use arduino, check out the teensy 3.

closed 2017-06-05 by corecode